maniacbug/RF24 #60: RF24 radio.printDetails() does not print anything on serial monitor

debojitk, 2015-01-16:
Hi, 
I tried the RF24 library but it is not printing the debug detail at serial monitor. could you please point me out what wrong i am doing? This issue took a whole day of my frustration. How would i know if my device is not dead, or setup is not wrong? 
I am trying the pingpair example, and the printf method working perfect, but the printdetails method is printing nothing. 
Anyway the setup is not working although. Could you please tell me if I remove the device at all, whether the printdetails should still work. I am asking this to isolate if the problem with the arduino itself or the problem with the device setup. If it does not print any detail if the device is not at all connection then i am sure that it is not printing for my case becuase of some wrong wiring. If it prints (although some garbage) something without the device connected, then it would be definitely some stdout redirection issue.
Just to add, I added a Serial.println as the first line inside the RF24Debug. printdetails() method, but this Serial.println does not work even. It seems the method is not being called altogether.
Could you please point out whats going wrong in my case?

Thanks in advance,
Debojit

ZonkerHarris, 2015-01-16:
Did you include the serial.begin in your setup? Is your bps rate the same in the serial.begin and the pseudo-console?

    -Z-

debojitk, 2015-01-16:
Hi,

My problem was the faulty library (it did not contain the getting started sample at all, and the culprint RF24Debug.cpp, i dont remember from where i got this library, very misleading though). I took the library from https://github.com/maniacbug/RF24 again, reinstalled the library, and it worked like a charm. I also added a 10uf cap across the vcc and gnd to improve stability.
One problem though: Initially while the radiodetail was printed it was showing a wrong address for first device (and for both the transceivers), while I started the transmission mode for both (with pressing T), then it showed the correct address for both the radio, and then i switched one end back to receiving mode.

Indoor test result: 30 ft with all wall/door obstacle - pretty good with 1 mbps speed, i will try with 250 kbps, to see if the range gets increased or not.

Thanks again
Debojit

closed 2015-01-16 by debojitk