maniacbug/RF24 #69: Unable to change MISO/MOSI/SCK Pins

shaiss, 2015-09-28:
I have looked through the code and cannot find where to change the MISO/MOSI/SCK pins from their default definitions:
  5 - SCK to Arduino pin 13
   6 - MOSI to Arduino pin 11
   7 - MISO to Arduino pin 12 

I'm using my NRF with a Funduino Joystick Shield, and they map the SPI differently.  per http://www.elecfreaks.com/wiki/index.php?title=Joystick_Shield#External_Links

D11	11	SCK
D12	12	MOSI
D13	13	MISO

Note that in order to make the library work with the Joystick shield, I need to figure out a way to flip these 3 pins.  Anyone have any idea where these pins are defined in the library?

FrankOcean11, 2015-09-28:
are you saying you read the cpp and header files and nothing?

Frank's Desig'Nature

"Designed Signature Inspired by Nature"

shaiss, 2015-09-28:
@frankocean11 I tried my best.  It's probably there, but I'm no pro so I may have missed something.

shaiss, 2015-09-28:
The pinout in the joystick wiki is wrong.  The library spi pins work as described

closed 2015-09-28 by shaiss

a20084666, 2015-09-28:
so  ga