maniacbug/RF24 #72: Problems to run example Scanner dont find WProgram.h

enrutador, 2016-03-06:
Hi and congratulations for the work
I run Scanner example but the code fails because dont find Wprogram.h:

Arduino:1.6.7 (Windows 7), Placa:"Arduino/Genuino Uno"
sketch\output\scanner.cpp:1:22: fatal error: WProgram.h: No such file or directory
 #include <WProgram.h>
compilation terminated.
exit status 1

I see this post but not solve the problem:

https://github.com/adafruit/DHT-sensor-library/issues/1

Thanks

wwin3286tw, 2016-03-13:
Try modify WProgram.h to Arduino.h

wwin3286tw, 2016-03-20:
http://forum.arduino.cc/index.php?topic=147680.0

jryx, 2016-03-20:
I'm having the same problem. There is no include for WProgram.h directly in the example sketch. I browsed around in the maze of included files but can't seem to find it. Do you know what file it's in?