maniacbug/RF24 #94: NRF24L01+ TWO WAY COMMUNICATION

sbmmkl, 2020-01-21:
hi,
I'm working on a project in which I want two nrf24l01+ modules to communicate with each other both of them working fine and transmitting data to each other and one of them is perfectly receiving data but the other one is not receiving or perfectly (1 in 100).
I have figured out that radio.setAutoAck(bool) makes some different.

mooymanh, 2020-01-21:
I have a similar project,  Did find out that the servomotor, powered with a separate battery needed to have its ground line also to be connected to the Arduino. However the project still does not work.
How did you find out that one of the two is perfectly receiving data?

JGM

sbmmkl, 2020-01-21:
> I have a similar project,  Did find out that the servomotor, powered with a separate battery needed to have its ground line also to be connected to the Arduino. However the project still does not work.
> How did you find out that one of the two is perfectly receiving data?
> JGM

I'm working with this module for 1 year but not all the time I want 2-way communication.
I got information from some random sites and data-sheet about how nRF24l01+ works and after that, I have decided to create 2 pipelines for 2-way communication and at the last, I got some code from youtube channels.
my 2-way communication is working fine but it not receiving data in fixed delay means when TX transmits Rx should ready to receive but in this case both are transmitting and receiving, that's why both receiving times are not fixed.
in the end, I have decided to use IRQ pin of nRF24l01+ as interrupter for microcontroller but I don't know how to set IRQ to change its state when receive data. on github i found 

> I found that MASK_TX_DS and MASK_MAX_RT these two bits must be declared as 1 and MASK_RX_DR this bit as 0 to enable receive only interrupt . Is it right ? and then how can I configure this bit ?

sbmmkl, 2020-01-21:
> I have a similar project,  Did find out that the servomotor, powered with a separate battery needed to have its ground line also to be connected to the Arduino. However the project still does not work.
> How did you find out that one of the two is perfectly receiving data?
> JGM

I'm working with this module for 1 year but not all the time I want 2-way communication.
I got information from some random sites and data-sheet about how nRF24l01+ works and after that, I have decided to create 2 pipelines for 2-way communication and at the last, I got some code from youtube channels.
my 2-way communication is working fine but it not receiving data in fixed delay means when TX transmits Rx should ready to receive but in this case both are transmitting and receiving, that's why both receiving times are not fixed.
in the end, I have decided to use IRQ pin of nRF24l01+ as interrupter for microcontroller but I don't know how to set IRQ to change its state when receiving data. on github i found 

> I found that MASK_TX_DS and MASK_MAX_RT these two bits must be declared as 1 and MASK_RX_DR this bit as 0 to enable receive only interrupt . Is it right ? and then how can I configure this bit ?

mooymanh, 2020-01-23:
I read Arduino Forum, Communty=> Exhibition/Gallery => Topic started by
Robin2 on August 28
The first page mentions problems with nRF24L01.   May be this helps.